probe-rs/cargo-flash #6: Show information about permissions / udev when no probe is found

Tiwalun, 2020-01-29:
To help first time users, we should include information about the most common problems when no probe is found.

For Linux and Daplink probes, this is usually a permission problem, and can be fixed by adding udev rules or changing permissions. The error message should either directly include this, or link to a web page explaining the setup.

Yatekii, 2020-01-29:
We could also provide a method to automatically install udev rules :) Of course the user would ahve to run with sudo and add a flag.

Yatekii, 2020-07-17:
This could also be eimplemented for https://github.com/probe-rs/cargo-embed.

Yatekii, 2022-01-29:
This was actually implemented with the new diagnostics in #159.

closed 2022-01-29 by Yatekii