probe-rs/cargo-flash #9: cargo-flash does not actually reset after flashing

therealprof, 2020-02-16:
**Describe the bug**

I'm flashing a binary to a Nucleo-F072RB like this:
```
# cargo flash --chip STM32F072RBTx --example flashing_irq
    Finished dev [unoptimized + debuginfo] target(s) in 0.02s
    Flashing /Users/egger/OSS/interrupt-comparison/target/thumbv6m-none-eabi/debug/examples/flashing_irq
    Erasing sectors   ✔ [00:00:00] [##################################################################################################################################]  12.00KB/ 12.00KB @  21.79KB/s (eta 0s )
    Programming pages ✔ [00:00:01] [##################################################################################################################################]  12.00KB/ 12.00KB @   5.28KB/s (eta 0s )
    Finished in 1.776s
```

However, unlike the help explains in the command help, it does not actually reset the target but halt instead:
```
        --reset-halt              Use this flag to reset and halt (instead of just a reset) the attached core after
                                  flashing the target.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Flash binary as mentioned above

**Expected behavior**
That the program runs automatically after flashing.

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**
 - macOS High Sierra

**Additional context**
None

Yatekii, 2020-02-16:
Thanks a lot for reporting :) This seems to be an issue with the core reset in the STM32s; I experienced this while debugging the ITM modules too.

According to my experiments resetting works as expected with Nordic MCUs.

We will have to investigate this more :)

Yatekii, 2020-02-28:
Newer discoveries have unveiled that this is indeed an issue for the nRF52840 too.

Works for:
- nRF51822
- LPC845

Fails for:
- nRF52840
- STM32F072
- STM32F407

Please add your discoveries too :)

dbrgn, 2020-02-28:
Fails for:

- STM32L031K6

closed 2020-03-16 by Yatekii