probe-rs/cargo-flash #20: Saving command line flags in .cargo/config

birkenfeld, 2020-04-01:
It would be nice to be able to store values for parameters such as `--chip` or `--speed` in the `.cargo/config` file, since they typically stay the same while working on a project.

birkenfeld, 2020-04-01:
Otherwise, great tool!

If you're curious, the reason I need to specify `speed` is that the "Chinese ST-Link clone" I'm using produces communication errors otherwise. OpenOCD seems to detect this automatically.

There are also about 20 identical timeout warnings while erasing, but it seems to work fine:
```
WARN probe_rs::flashing::flasher > Error while waiting for core halted: An error with the usage of the probe occured: Operation timed out.
```

Yatekii, 2020-04-01:
Hey! Glad you like the tool!

Yes, config files are definitely a planned thing, it's just too many things and too few time :D This is definitely an easy issue a new contributor could take on :)

Hmm could you tell us what ST-Link it is exactly and with what speed it works and with which it doesn't? Could you maybe make an issue for this in the main repo https://github.com/probe-rs/probe-rs? Thanks!

Yeah, so basically ARM cores can let you wait for quite a bit. So we have to choose a timeout to not overdo it. If we timeout we try again at some places. Maybe we should make the timeouts longer so not as many warings appear.

Maybe you can give us your exact setup?

Thanks for reporting =)

adamgreig, 2020-04-08:
> There are also about 20 identical timeout warnings while erasing, but it seems to work fine:
> 
> ```
> WARN probe_rs::flashing::flasher > Error while waiting for core halted: An error with the usage of the probe occured: Operation timed out.
> ```

I've fixed this spurious error in https://github.com/probe-rs/probe-rs/pull/221 by silencing only the timeout variant, which was also triggering when I sped up accesses in that PR. If it really times out, the outer loop is what catches it and raises a real timeout error.

Yatekii, 2020-07-17:
So selecting the chip via metadata was actually implemented in #31 and for all other needs I would like to kindly point you to https://github.com/probe-rs/cargo-embed which allows for far more extensive settings :)

I would like to keep the feature creep on cargo-flash really low and put everything larger into cargo-embed :)

closed 2020-07-17 by Yatekii