probe-rs/cargo-flash #21: cargo flash does not use target in `.cargo/config` when `--target` is omitted

SamP20, 2020-04-01:
**Bug description**
Cargo flash does not appear to use the target defined in `.cargo/config` when `--target` is omitted. It looks in `target/release` rather than `target/thumbv7em-none-eabihf/release` for the executable to flash.

**To Reproduce**
Steps to reproduce the behavior:
1. Using the code in this repository https://github.com/SamP20/rust-getting-started
2. Run `cargo flash --release --chip stm32f439zitx`
3. The output is
    ```
        Finished release [optimized + debuginfo] target(s) in 0.04s
        Flashing \\?\C:\<snip>\rust-getting-started\target\release\rust-getting-started
     failed to flash \\?\C:\<snip>\rust-getting-started\target\release\rust-getting-started: The system cannot find the file specified. (os error 2)
    ```

**Expected behavior**
Cargo flash should use the target defined in `.cargo/config` when `--target` is omitted, and the output should include `Flashing \\?\C:\<snip>\rust-getting-started\target\thumbv7em-none-eabihf\release\rust-getting-started`

**Desktop**
Windows 10 (using powershell)

**Any additional information**
Using cargo flash version 0.6.0

Yatekii, 2020-04-01:
Hey thanks for reporting this!

This is odd. This works perfectly fine on Linux. Or at least I have never heard about this not working. Maybe [cargo-project](https://crates.io/crates/cargo-project) does not work as expected on windows?

Will have to investigate.

SamP20, 2020-04-01:
I have also noticed that https://github.com/rust-embedded/cargo-binutils/issues/32 has regressed, so perhaps it is an upstream issue

Yatekii, 2020-04-01:
Well, but why is it no problem on Linux? Also, cargo-project didn't update and the last dep of it that did is serde and toml. So that would be serious regression in those 2 ...

SamP20, 2020-04-01:
I think I've found the issue here https://github.com/japaric/cargo-project/blob/0ff369198537a6ae361c0674f35594761551eaed/src/lib.rs#L79. There's a hardcoded forward slash in `".cargo/config"`. When searching for the file `Path::exists()` returns false because for some reason the forward slash isn't recognised (I'm a bit surprised about this as Windows usually handles forward slashes fine)

If I replace that string with `let cargo_config = Path::new(".cargo").join("config");` and make the appropriate changes elsewhere then it seems to work.

If you're happy with that then I'll go ahead and submit a PR upstream

Yatekii, 2020-04-01:
Nice, sounds great! Stupid windows and their paths ...

Would be great if you PR that and ping japaric for a hotfix release =)

We need some more fixes to cargo-project regarding workspaces too ... but that's for different times :D

elfmimi, 2020-04-27:
for cross reference: https://github.com/japaric/cargo-project/pull/8#issuecomment-619929683

elfmimi, 2020-05-29:
cargo-project v0.2.4 is out. just adopting it will fix this issue. @Yatekii

Yatekii, 2020-05-29:
Thanks for the hint! Installing cargo-flash via cargo will automatically take the new version. I am going to close this :)

closed 2020-05-29 by Yatekii