probe-rs/cargo-flash #22: Failed to build on Arch Linux (x86_64)

yusefkarim, 2020-04-02:
I tried installing cargo-flash via Cargo and am getting a compile failure due to the hidapi v1.2.1 dependency due to libusb. Below is the error output:

RUST_BACKTRACE=1 cargo install cargo-flash
```sh
error: failed to run custom build command for `hidapi v1.2.1`

Caused by:
  process didn't exit successfully: `/tmp/cargo-installR8kIYb/release/build/hidapi-cee524790ccb98ec/build-script-build` (exit code: 101)
--- stderr
thread 'main' panicked at 'Unable to find libusb-1.0: "Failed to run `\"pkg-config\" \"--libs\" \"--cflags\" \"libusb-1.0\"`: No such file or directory (os error 2)"', /home/cuyler/.cargo/registry/src/github.com-1ecc6299db9ec823/hidapi-1.2.1/build.rs:65:21
stack backtrace:
   0: backtrace::backtrace::libunwind::trace
             at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/libunwind.rs:88
   1: backtrace::backtrace::trace_unsynchronized
             at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/mod.rs:66
   2: std::sys_common::backtrace::_print_fmt
             at src/libstd/sys_common/backtrace.rs:77
   3: <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt
             at src/libstd/sys_common/backtrace.rs:59
   4: core::fmt::write
             at src/libcore/fmt/mod.rs:1052
   5: std::io::Write::write_fmt
             at src/libstd/io/mod.rs:1426
   6: std::sys_common::backtrace::_print
             at src/libstd/sys_common/backtrace.rs:62
   7: std::sys_common::backtrace::print
             at src/libstd/sys_common/backtrace.rs:49
   8: std::panicking::default_hook::{{closure}}
             at src/libstd/panicking.rs:204
   9: std::panicking::default_hook
             at src/libstd/panicking.rs:224
  10: std::panicking::rust_panic_with_hook
             at src/libstd/panicking.rs:472
  11: rust_begin_unwind
             at src/libstd/panicking.rs:380
  12: core::panicking::panic_fmt
             at src/libcore/panicking.rs:85
  13: core::option::expect_none_failed
             at src/libcore/option.rs:1199
  14: build_script_build::compile_linux::{{closure}}
  15: build_script_build::main
  16: std::rt::lang_start::{{closure}}
  17: std::rt::lang_start_internal::{{closure}}
             at src/libstd/rt.rs:52
  18: std::panicking::try::do_call
             at src/libstd/panicking.rs:305
  19: __rust_maybe_catch_panic
             at src/libpanic_unwind/lib.rs:86
  20: std::panicking::try
             at src/libstd/panicking.rs:281
  21: std::panic::catch_unwind
             at src/libstd/panic.rs:394
  22: std::rt::lang_start_internal
             at src/libstd/rt.rs:51
  23: main
  24: __libc_start_main
  25: _start
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

warning: build failed, waiting for other jobs to finish...
error: failed to compile `cargo-flash v0.6.0`, intermediate artifacts can be found at `/tmp/cargo-installR8kIYb`

Caused by:
  build failed
```
I currently have libusb 1.0.23-2 installed and have read Arch includes -dev files with the main package (there is no libusb-dev*). I have also tried installing libusb-compat 0.1.7-1 and libusb0 (from AUR) and had no luck.

Let me know if I can provide anymore detail or help out in anyway. Thank you!

Tiwalun, 2020-04-02:
Hi, thanks for reporting this!

Please try just running pkg-config itself:

    pkg-config --libs --cflags libusb-1.0

This should show an ouput like this (on my Ubuntu machine):

    -I/usr/include/libusb-1.0 -lusb-1.0

yusefkarim, 2020-04-02:
Hi @Tiwalun,

Reading your response made me realize that pkg-config is an external command and I just didn't have it installed, I had never heard of pkg-config and thought it was part of Cargo or something!

On Arch Linux I did:
```sh
pacman -S pkgconf
```
Running the command you suggested works as intended now:
```sh
pkg-config --libs --cflags libusb-1.0
```
```
-I/usr/include/libusb-1.0 -lusb-1.0
```
and ```cargo install cargo-flash``` works now of course.

Thank you very much for the response!

closed 2020-04-02 by yusefkarim