probe-rs/cargo-flash #27: USB error while opening USB device: Entity not found

FluenTech, 2020-04-12:
**Describe the bug**
```
> cargo.exe +nightly flash --chip nrf52832
    Finished dev [unoptimized + debuginfo] target(s) in 0.16s
    Flashing \\?\C:\<clipped>\target\debug\rust-cross
       Error An error specific to a probe type occured: USB error while opening USB device: Entity not found

Process finished with exit code 1
```
I have also tried various arguments such as the explicit target `--target thumbv7em-none-eabihf` and probe index `--probe-index 0` (I only have the one J-Link connected). All of these attempts resulted in the exact same output.

**Desktop**
 - Windows 10

**Additional context**
- Using the nRF52 DK with SEGGER J-Link OB
- Running the cargo flash command from Powershell

Yatekii, 2020-04-12:
Thanks for reporting this! I think there is some specific driver required on windows. @Tiwalun might be able to help you. And quite possibly @jonas-schievink :)

Tiwalun, 2020-04-20:
probe-rs doesn't work with the offical J-Link drivers on Windows, unfortunately. You need to replace the official driver with the WinUSB driver. An easy way to do this is by using [Zadig](https://zadig.akeo.ie/). 

Please note that you will not be able to use the normal Segger J-Link tools if you replace the driver.

PTaylor-FluenTech, 2020-04-20:
Thank you. That answers my question.

closed 2020-04-20 by PTaylor-FluenTech