probe-rs/cargo-flash #36: Suddenly unable to flash to `STM32F407VGTx` (STM32F407G-DISC1 ST-LINK V2.1)

mitchmindtree, 2020-07-17:
Thanks for your work on `cargo-flash`!

## The Bug

After (successfully) flashing a particular example onto my STM32F407G-DISC1 last night, `cargo-flash` no longer seems able to attach itself successfully to my board - as a result, I can't seem to flash a new program onto it.

## Details

Last night I was trying some different programs on my new STM32F407G-DISC1 using `cargo flash`. I was able to edit and flash the examples at the [stm32f407g-disc1 board support crate](https://github.com/stm32-rs/stm32f407g-disc) without any issues. Things were looking great! Here's an example of the command I was using:

```
cargo flash --release --example gpio_hal_blinky --chip STM32F407VGTx
```

I then tried adapting the [`rtic-examples` bluepill example](https://github.com/rtic-rs/rtic-examples/tree/master/rtic_v5/bluepill_blinky) to see if I could blink the stm32 crate with blinky. See [here for the adapted example](https://github.com/mitchmindtree/stm32f4xx-rtic-blinky) that I flashed onto the board with the following command:

```
cargo flash --release --chip STM32F407VGTx
```

Everything appeared to work nicely! Here's the output of the seemingly successful flash:

```
    Finished release [optimized + debuginfo] target(s) in 1m 07s                                                                              
    Flashing /home/mindtree/programming/rust/stm32f4xx-rtic/target/thumbv7em-none-eabihf/release/stm32f4xx-rtic                               
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Request), new reason is Exception     
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Breakpoint), new reason is Request    
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Request), new reason is Exception     
     Erasing sectors ✔ [00:00:00] [#################################################################]  16.00KB/ 16.00KB @  37.44KB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [#################################################################]   4.00KB/  4.00KB @   3.47KB/s (eta 0s )
```

I can see that the example appears to run as expected, blinking the LED each time the `blink` software task gets called.

However, any of the following times I have attempted to flash a program `cargo-flash` gives a `JtagNoDeviceConnected` error. The first time I encountered the error, the output looked like this:

<details>
  <summary><b>Error output</b></summary>
  
  stdout+stderr:
  
  ```
   Compiling stm32f4xx-rtic v0.1.0 (/home/mindtree/programming/rust/stm32f4xx-rtic)                                                           
    Finished release [optimized + debuginfo] target(s) in 0.75s                                                                               
    Flashing /home/mindtree/programming/rust/stm32f4xx-rtic/target/thumbv7em-none-eabihf/release/stm32f4xx-rtic                               
        WARN probe_rs::probe::stlink > check_status failed: JtagNoDeviceConnected
       Error failed attaching to target

Caused by:
    0: An error with the usage of the probe occured
    1: An error specific to a probe type occured
    2: Command failed with status JtagNoDeviceConnected

Stack backtrace:
   0: cargo_flash::main_try
   1: cargo_flash::main
   2: std::rt::lang_start::{{closure}}
   3: std::rt::lang_start_internal::{{closure}}
             at src/libstd/rt.rs:52 
      std::panicking::try::do_call
             at src/libstd/panicking.rs:297
      std::panicking::try
             at src/libstd/panicking.rs:274
      std::panic::catch_unwind
             at src/libstd/panic.rs:394
      std::rt::lang_start_internal
             at src/libstd/rt.rs:51 
   4: main
   5: __libc_start_main
   6: _start
[... 199 lines cut ...]
```rust
        // Setup clocks
        let rcc = device.RCC.constrain();
        let _clocks = rcc
            .cfgr
            .use_hse(8.mhz())
            .sysclk(72.mhz())
            .pclk1(36.mhz())
            .freeze();
```

This config is left over from the bluepill config, I managed to miss this when updating the code! I think likely it should have looked like this:
```rust
        let rcc = device.RCC.constrain();
        let _clocks = rcc.cfgr.sysclk(168.mhz()).freeze();
```

Is it possible that my odd clock settings might have caused some issues with the debug interface `cargo-flash` tries to use? Or do you think something else might be going on? Any advice on other approaches to try would be appreciated!

## System Info

cargo-flash 0.8.0
rustc 1.46.0-nightly (feb3536eb 2020-06-09)
cargo 1.45.0-nightly (40ebd5220 2020-06-01)
nixos, linux 5.7.2

mitchmindtree, 2020-07-17:
Possibly related: probe-rs/probe-rs#216

mitchmindtree, 2020-07-17:
After forcing the STM32F407G-DISC1 to stay in the bootloader on reset by setting the BOOT0 pin high (connecting it to VDD), I was then able to successfully `cargo flash` the device again!

I'll leave this open in case you'd still like to see this addressed.

adamgreig, 2020-07-17:
Are you able to reproduce the issue again if you re-flash the bad firmware? And if so, while it's failing with cargo-flash, are you able to connect or reprogram using other tools, such as st-util, openocd, or pyocd? From a quick look at  your firmware nothing except the HSE part seems suspicious. I think by default the HSE on that board is 12MHz, so you might have been overclocking your board 50% which could have caused the issue. Perhaps you could see if `use_hse(12.mhz())` makes a difference?

mitchmindtree, 2020-07-17:
Hi @adamgreig!

Yes I can recreate the issue by flashing the bad firmware again.

I changed the clock setup to:

```rust
        let rcc = device.RCC.constrain();
        let _clocks = rcc.cfgr.sysclk(168.mhz()).freeze();
```

But I still run into the same issue. Here is the generated `target/rtic-expansion.rs` in case it you're curious to see what gets generated:

<details>
  <summary><b>rtic-expansion.rs</b></summary>
  
  Generated code:

```rust
#[allow(non_snake_case)]
fn init(cx: init::Context) -> init::LateResources {
    let mut core = cx.core;
    core.DWT.enable_cycle_counter();
    let device: stm32f4xx_hal::stm32::Peripherals = cx.device;
    let rcc = device.RCC.constrain();
    let _clocks = rcc.cfgr.sysclk(CYCLE_HZ.hz()).freeze();
    let gpiod = device.GPIOD.split();
    let mut led = gpiod.pd13.into_push_pull_output();
    led.set_low().unwrap();
    cx.schedule.blink(cx.start + CYCLE_HZ.cycles()).unwrap();
    init::LateResources { led }
}
#[allow(non_snake_case)]
fn blink(cx: blink::Context) {
    use rtic::Mutex as _;
    cx.resources.led.toggle().unwrap();
    cx.schedule.blink(cx.scheduled + CYCLE_HZ.cycles()).unwrap();
}
#[doc = r" Resources initialized at runtime"]
#[allow(non_snake_case)]
pub struct initLateResources {
    pub led: PD13<Output<PushPull>>,
}
#[allow(non_snake_case)]
#[doc = "Initialization function"]
pub mod init {
    #[doc = "Tasks that can be `schedule`-d from this context"]
    #[derive(Clone, Copy)]
    pub struct Schedule {
        _not_send: core::marker::PhantomData<*mut ()>,
    }
    #[doc(inline)]
    pub use super::initLateResources as LateResources;
    #[doc = r" Execution context"]
    pub struct Context {
        #[doc = r" System start time = `Instant(0 /* cycles */)`"]
        pub start: <rtic::cyccnt::CYCCNT as rtic::Monotonic>::Instant,
        #[doc = r" Core (Cortex-M) peripherals minus the SysTick"]
        pub core: rtic::Peripherals,
        #[doc = r" Device peripherals"]
        pub device: stm32f4xx_hal::stm32::Peripherals,
        #[doc = "Tasks that can be `schedule`-d from this context"]
        pub schedule: Schedule,
    }
    impl Context {
        #[inline(always)]
        pub unsafe fn new(core: rtic::Peripherals) -> Self {
            Context {
                start: <rtic::cyccnt::CYCCNT as rtic::Monotonic>::zero(),
                device: stm32f4xx_hal::stm32::Peripherals::steal(),
[... 302 lines cut ...]
        );
        rtic::export::NVIC::unmask(stm32f4xx_hal::stm32::Interrupt::EXTI0);
        let _ = [(); ((1 << stm32f4xx_hal::stm32::NVIC_PRIO_BITS) - 1u8 as usize)];
        core.SCB.set_priority(
            rtic::export::SystemHandler::SysTick,
            rtic::export::logical2hw(1u8, stm32f4xx_hal::stm32::NVIC_PRIO_BITS),
        );
        core.SYST
            .set_clock_source(rtic::export::SystClkSource::Core);
        core.SYST.enable_counter();
        core.DCB.enable_trace();
        core.SCB.scr.modify(|r| r | 1 << 1);
        let late = crate::init(init::Context::new(core.into()));
        led.as_mut_ptr().write(late.led);
        <rtic::cyccnt::CYCCNT as rtic::Monotonic>::reset();
        rtic::export::interrupt::enable();
        loop {
            rtic::export::wfi()
        }
    }
};
```
</details>

I haven't tried reprogramming with other tools yet - I'll have a go using st-util now and report back.

mitchmindtree, 2020-07-17:
OK, I haven't tried any other tools yet, but I've managed to get the example to work!

By declaring an `rtic` `#[idle]` task, the issue no longer occurs and I can reprogram the board using `cargo flash` without issues. Here's the `idle` function I've used:

```rust
    #[idle]
    fn idle(_: idle::Context) -> ! {
        loop {
            core::sync::atomic::spin_loop_hint();
        }
    }
```

I wonder why this might be the case?

adamgreig, 2020-07-17:
Oh, of course, it's because the [default idle](https://github.com/rtic-rs/cortex-m-rtic/blob/6e233bec029810716c5a48113b22f4446204cfab/macros/src/codegen/idle.rs#L87) in RTIC calls `WFI` which puts the device to sleep, from which it can't be accessed by the debugger unless you first configure "debug in sleep" mode. Putting your own idle is probably the way to go. I would have expected that connecting under reset in probe-rs should fix this though.

Yatekii, 2020-07-17:
Yeah, I believe that this is still not fully fixed.

Maybe I should flash a WFI example onto one of my kits to debug this! We'll see during oxidize :)