probe-rs/cargo-flash #42: cargo flash does not seem to respect .cargo/config.toml

jounathaen, 2020-08-09:
Apparently cargo flash does only respect target settings from `.cargo/config`. I heard that the cool-kids now [specify the target in `.cargo/config.toml`](https://doc.rust-lang.org/cargo/reference/config.html) but cargo flash seems to ignore this file as of v0.8.0.

Thanks for providing this cool project! :+1:

jounathaen, 2020-08-09:
Related: https://github.com/probe-rs/cargo-embed/issues/29

Yatekii, 2020-08-09:
Thanks for reporting this! This is something that is handled by cargo-project which we have no control over strictly speaking. Ofc someone could make a PR.

@Tiwalun has submitted a PR (https://github.com/probe-rs/cargo-flash/pull/41) that should improve the state here as it will use cargo-metadata in the future. Could you try if this PR helps? :)

jounathaen, 2020-08-09:
Yeah, the code from the PR would solve this issue. (Haven't tested it otherwise...)
```
> /tmp/cargo-flash/target/release/cargo_flash --chip STM32F411CEUx --protocol swd
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
    Flashing /xxx/target/thumbv7em-none-eabihf/debug/blinky
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Request), new reason is Exception
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Breakpoint), new reason is Request
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Request), new reason is Exception
     Erasing sectors ✔ [00:00:00] [##################################]  32.00KB/ 32.00KB @  41.87KB/s (eta 0s )
 Programming pages   ✔ [00:00:01] [##################################]  21.00KB/ 21.00KB @  10.21KB/s (eta 0s )
```

Yatekii, 2020-08-09:
Great :) The PR will be merged soon and should also find it's way into cargo-embed :)

Tiwalun, 2020-08-10:
Fixed by #41

closed 2020-08-10 by Tiwalun