probe-rs/cargo-flash #52: cargo flash does not parse .cargo/config.toml (expects only .cargo/config)

andresovela, 2020-08-14:
It causes some issues, like failing use the appropriate build target.

**To Reproduce**
Steps to reproduce the behavior:
Rename `.cargo/config` to `.cargo/config.toml`

**Expected behavior**
The file should be parsed.
**Desktop (please complete the following information):**
 - macOS Catalina 10.15.5

andresovela, 2020-08-14:
I heard the argument that the file *must* be called just `config`, but the Cargo book says otherwise:

> **Note**: Cargo also reads config files without the `.toml` extension, such as `.cargo/config`. Support for the `.toml` extension was added in version 1.39 and is the preferred form. If both files exist, Cargo will use the file without the extension.

Source: https://doc.rust-lang.org/cargo/reference/config.html

Yatekii, 2020-08-14:
This is already fixed and will be in the next release :)

closed 2020-08-14 by Yatekii