probe-rs/cargo-flash #79: Compilation Error / Cargo-Install Fails

rincewound, 2020-09-14:
using `cargo install cargo-flash` downloads  (presumably) all dependant crates and starts compiling everything. As soon as it gets to the linking stage the following error occurs:
error: linking with `link.exe` failed: exit code: 1181

(...)
 = note: Non-UTF-8 output: LINK : fatal error LNK1181: Eingabedatei \"ftdi1.lib\" kann nicht ge\x94ffnet werden.\r\n

error: aborting due to previous error

error: failed to compile `cargo-flash v0.9.0`

My toolchain is:
stable-x86_64-pc-windows-msvc (default)
rustc 1.45.0 (5c1f21c3b 2020-07-13)

Tiwalun, 2020-09-14:
Unfortunately, not all dependencies are available as crates yet.

On Windows, you need to install libftdi using vcpkg. See https://github.com/probe-rs/cargo-flash#windows for instructions.

rincewound, 2020-09-14:
I get the same error with cargo-embed.

rincewound, 2020-09-14:
Thanks a lot, I'll check that out

Tiwalun, 2020-09-14:
No problem, the instructions are still missing from the crates.io page. 

We should probably fix that.

rincewound, 2020-09-14:
FYI: If you've installed multiple versions of Visual Studio next to each other you will wake up in a world of pain, when trying to install/compile libftdi and libusb, details are here:
https://stackoverflow.com/questions/14372706/visual-studio-cant-build-due-to-rc-exe

stappersg, 2020-09-25:
Here, Debian Bullseye, something similar:
```
note: /usr/bin/ld: cannot find -lftdi1
```

And as https://github.com/probe-rs/cargo-flash#linux  advices / says:
```
sudo apt install -y libftdi1-dev
```

Thing I advice / say

Webpage https://probe.rs/guide/1_tools/cargo-flash/  should include  _build requirements_.

Yatekii, 2020-11-08:
The requirement for libftdi has been hidden behind a feature flag :) So in the upcomming v0.10.0 this wont be of issue anymore :) I'll leave this issue open as a reminder to add this in some form to the website, is this ok for you @stappersg  & @rincewound ?

rincewound, 2020-11-16:
sure :)

ungerik, 2020-12-10:
On Ubuntu 20.04 cargo install fails because this Ubuntu version has no `libusb-1.0`.
Instead, there is `libusb-1.0-0`.

ungerik, 2020-12-10:
OK, this fixed my problem on Ubuntu 20.04, so just updating the readme would be sufficient:
`sudo apt-get install libudev-dev libusb-1.0-0-dev`

Tiwalun, 2020-12-10:
Thanks for the report!

It seems that our CI still uses version 18.04, where the libusb-dev package was sufficient.

Are you sure libudev-dev is necessary? It seems to work on Ubuntu 20.04 with just `libusb-1.0-0-dev` (See probe-rs/probe-rs#478).

ungerik, 2020-12-10:
I didn't test without `libudev-dev`, but there is no special reason I included it other than I copy-pasted the command line from another project and it worked after that.

Yatekii, 2021-05-11:
Hmm can we close this?

closed 2021-05-23 by Tiwalun