probe-rs/cargo-flash #153: failed to run custom build command for `miniz_oxide v0.4.3`

gyzhangqi, 2021-01-03:
**Describe the bug**
 failed to compile `cargo-flash v0.10.2`, intermediate artifacts can be found at `C:\Users\XXXXX\AppData\Local\Temp\cargo-installAgPc8Q`.

**To Reproduce**
Steps to reproduce the behavior:
1. >cargo install cargo-flash
2. then:
Updating crates.io index
  Installing cargo-flash v0.10.2
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling syn v1.0.57
   Compiling winapi v0.3.9
   Compiling autocfg v1.0.1
   Compiling serde_derive v1.0.118
   Compiling serde v1.0.118
   Compiling libc v0.2.81
   Compiling crc32fast v1.2.1
   Compiling cfg-if v1.0.0
   Compiling cc v1.0.66
   Compiling pkg-config v0.3.19
   Compiling version_check v0.9.2
   Compiling lazy_static v1.4.0
   Compiling ucd-trie v0.1.3
   Compiling adler32 v1.2.0
   Compiling log v0.4.11
   Compiling memchr v2.3.4
   Compiling rle-decode-fast v1.0.1
   Compiling libflate_lz77 v1.0.0
   Compiling linked-hash-map v0.5.3
   Compiling maplit v1.0.2
   Compiling bitflags v1.2.1
   Compiling vcpkg v0.2.11
   Compiling cfg-if v0.1.10
   Compiling dtoa v0.4.7
   Compiling unicode-width v0.1.8
   Compiling ryu v1.0.5
   Compiling adler v0.2.3
   Compiling byteorder v1.3.4
   Compiling regex-syntax v0.6.21
   Compiling base64 v0.13.0
   Compiling serde_json v1.0.61
   Compiling unicode-segmentation v1.7.1
   Compiling anyhow v1.0.37
   Compiling hashbrown v0.9.1
   Compiling proc-macro-hack v0.5.19
   Compiling fallible-iterator v0.2.0
   Compiling encode_unicode v0.3.6
   Compiling vec_map v0.8.2
   Compiling plain v0.2.3
   Compiling itoa v0.4.7
   Compiling strsim v0.8.0
   Compiling wasmparser v0.57.0
   Compiling stable_deref_trait v1.2.0
   Compiling svg v0.8.0
   Compiling number_prefix v0.3.0
   Compiling bitfield v0.13.2
   Compiling ihex v3.0.0
   Compiling humantime v2.0.1
   Compiling dunce v1.0.1
   Compiling thread_local v1.0.1
   Compiling num-traits v0.2.14
   Compiling miniz_oxide v0.4.3
   Compiling num-integer v0.1.44
   Compiling indexmap v1.6.1
   Compiling pest v2.1.3
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling yaml-rust v0.4.4
   Compiling textwrap v0.11.0
   Compiling heck v0.3.2
   Compiling hidapi v1.2.5
error: failed to run custom build command for `miniz_oxide v0.4.3`

Caused by:
  could not execute process `C:\Users\XXXXX\AppData\Local\Temp\cargo-installAgPc8Q\release\build\miniz_oxide-36cae9b9e6809b8e\build-script-build` (never executed)

Caused by:
  另一个程序正在使用此文件，进程无法访问。 (os error 32)
warning: build failed, waiting for other jobs to finish...
error: failed to compile `cargo-flash v0.10.2`, intermediate artifacts can be found at `C:\Users\XXXXX\AppData\Local\Temp\cargo-installAgPc8Q`

**Desktop**
 - Windows 10 professional 10.0.18363 - 1909

Yatekii, 2021-01-04:
Thanks for reporting this. It seem that you use a, I assume, Chinese locale. Is that correct?

The issue you are encountering is a problem with `miniz_oxide` which we don't really have control over. The package is used by `object` which we need for ELF parsing.

Would you mind filing an issue over here: https://github.com/Frommi/miniz_oxide ?

Yatekii, 2022-01-29:
Since there was never a followup and we have not had any more reports on this I am going to close this as we cannot really do anything about it even if it was still an issue. Please feel free to reopen if this still is an issue.

closed 2022-01-29 by Yatekii