probe-rs/cargo-flash #154: Add DFU support for USB and RS232

MarkuBu, 2021-01-07:
Most ARM microcontroller have a DFU bootloader over USB or USART. It would be great if cargo-flash would support this.

With DFU you would be able to flash a controller without the need for extra hardware.

JackRedstonia, 2021-02-21:
Seconding this, I possess a Blue Pill (STM32F103C8T6) with a DFU bootloader, which I flashed on using a cheap FTDI breakout clone. Purchasing an ST-Link/J-Link is not an option for me, unfortunately, so being able to program it with `cargo flash` with no extra hardware would be nice.

agausmann, 2021-02-23:
I believe this would be implemented as a probe in [probe-rs](https://github.com/probe-rs/probe-rs)? I'm not familiar with the internals, but I'd like to try to tackle this as it affects me too.

Yatekii, 2021-02-24:
Hi fellas!

It would certainly be cool to have this possibility in cargo-flash :)
It is NOT possible to implement a probe for this in probe-rs currently, as DFU really is no probe.

Maybe we could make a new crate which is specifically tailored ro DFU.

Maybe @Tiwalun has some ideas too :)

We'll definitely be here answering questions if you have any!

Disasm, 2021-02-24:
As an alternative solution, one can set a `runner` parameter in `.cargo/config` to point it to a custom script which runs objcopy and dfu-util to program the chip. This runner is triggered by `cargo run`.

Here is the example (not for DFU, but almost the same) to get the idea: [.cargo/config](https://github.com/Disasm/icebreaker-rs/blob/4a639cbe11dae02f34e64e6a3819577a102c8681/firmware/.cargo/config#L2), [flash.sh](https://github.com/Disasm/icebreaker-rs/blob/4a639cbe11dae02f34e64e6a3819577a102c8681/firmware/.cargo/flash.sh).