probe-rs/cargo-flash #167: Unable to find libusb-1.0: "Failed to run `\"pkg-config\" \"--libs\" \"--cflags\" \"libusb-1.0\"`: No such file or directory (os error 2)

lucaszanella, 2021-04-27:
On Ubuntu Focal (dockerized) with libusb-1-0-0-dev installed through apt-get, running `cargo install cargo-flash`:

```
error: failed to run custom build command for `hidapi v1.2.5`

Caused by:
  process didn't exit successfully: `/tmp/cargo-installm1Nsq2/release/build/hidapi-29605576d5e4594f/build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=LIBUSB_1.0_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUSB_1.0_STATIC
  cargo:rerun-if-env-changed=LIBUSB_1.0_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  thread 'main' panicked at 'Unable to find libusb-1.0: "Failed to run `\"pkg-config\" \"--libs\" \"--cflags\" \"libusb-1.0\"`: No such file or directory (os error 2)"', /root/.cargo/registry/src/github.com-1ecc6299db9ec823/hidapi-1.2.5/build.rs:65:60
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Did anyone get this erro also?

Tiwalun, 2021-04-27:
Do you have `pkg-config` installed?

lucaszanella, 2021-04-27:
Indeed it was the problem, thanks

closed 2021-04-27 by lucaszanella