probe-rs/cargo-flash #183: Chip field in manifest metadata unused when building flash loader

tmplt, 2021-07-05:
I'm refactoring some common functionality from `cargo-flash` into `probe-rs-cli-util` and noticed that the `package.metadata.chip` from the firmware manifest is only written to `METADATA` if found. When building the flash loader, only `--chip` is used.

Below displays the full scope-usage of `chip`:

https://github.com/probe-rs/cargo-flash/blob/0615fd5519c80cd1a7379745230798765978f1fa/src/main.rs#L360-L367

And when the flash loader is built:

https://github.com/probe-rs/cargo-flash/blob/0615fd5519c80cd1a7379745230798765978f1fa/src/main.rs#L414-L426

I presume this is a regression from #31?

tmplt, 2021-07-29:
When #188 merges this issue moves upstream to `probe-rs-cli-util::common_options`.

Yatekii, 2022-01-29:
@Tiwalun what do you think, should we implement this in `probe-rs-cli-util` or just in cargo-flash?