probe-rs/cargo-flash #198: Cannot read YAML from target-gen for Ambiq Micro

gauteh, 2021-10-11:
Hi,

I downloaded the pack file for Ambiq Apollo at https://developer.arm.com/tools-and-software/embedded/cmsis/cmsis-packs, and ran target-get pack to generate a yaml. Running:

cargo flash --chip-description-path out/Apollo\ Series.yaml --list-chips

gives:

```
       Error 0: Deserializing the yaml encountered an error
             1: flash_algorithms: invalid type: sequence, expected a map at line 245 column 3
```

gauteh, 2021-10-11:
Tried to add the yaml to probe-rs, seems to compile, but don't know if that works. But cargo-flash doesn't compile with that version. If I try to checkout probe-rs v0.11 and compile against that, I get the same yaml error when building.

Yatekii, 2022-01-29:
I believe this has been sorted on matrix :)

Please feel free to reopen if this is still an issue :)

closed 2022-01-29 by Yatekii