probe-rs/cargo-flash #211: Unable to compile  v0.12.0 and newer on Ubuntu 21.10 Impish

stillinbeta, 2021-12-11:
**Describe the bug**

I get a really long linker error

<details>j
<summary> output of `$ git checkout v0.12.0 && cargo build` </summary>

```
   Compiling cargo-flash v0.12.0 (/home/ellie/Projects/cargo-flash)
error: linking with `cc` failed: exit status: 1
  |
  = note: "cc" "-m64" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.13f62zdl6ye4yskv.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.13pmg7izfxbapl5w.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.13xbu3npk3e1rjnr.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1468sr8q74v83rtm.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.16vzn8gi8ojh3npp.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.17mz3pm4wq0tea10.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.17nif3m0e6spopsr.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1db6nxds2y90m1f1.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1ial00t2o0q3iov4.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1is8n0u1a2d3x5xr.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1jkdc85qz3bi66eg.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1k8yce84yvla9b76.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1lt3v9vhmxjbilrb.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1pebb06egk59k6tn.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1pw7isxobiwoncql.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1svmjk5m4x9wgcr5.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1u08i1qlx2gdhn92.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1vm2w6t9u7wyg02d.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.1wxo2mxozu3lo8re.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.21p1w9yk9locvwq3.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.25g2uaoaf4tkkr8a.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.269t0jxzzz5ui3hn.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.27def22lso57j8xv.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.2a5wntcxe1wjmmdj.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.2c6em5tm7hu4bt65.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.2j6ucmgjgz6rk6wc.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.2lga3kc14q5pmpxt.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.2ok340hnamemw4sl.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.2vrwwzns0bhemfdu.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.2wkoikyjpi86pu8l.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.2xwo3hez84f1w399.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.328mfn2f40v7aqau.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.33180myzgwtvxfnq.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.33yjfyvzgkm0pmwd.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.35ep0jfysj1gt9hu.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.361ml0s9k0baohlr.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.362rxrazfkdfhyh1.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.37xp7duutdf25g4z.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.3817cn45oq40h7op.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.383lyczhahl22th1.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.38j47l9546e7eom7.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.39s11ywpzf58h8dn.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.3c01bx91gongjwou.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.3dmkh64jcckhmp89.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.3f1nkpodyrsou1xl.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.3nzmd4geqkcuk8pi.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.3p3li4z8x55c4sl7.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.3q9fhie5ldazjq6m.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.3u30zz93ywjwcz3d.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.3uamd3ie43ouxs7u.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.3x30jbnrcoh3ixpu.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.3y3sbnoxtr5kdt1g.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.404cd1garbkufi5g.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.43g0g7wlfvgb9foj.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.45c7rb0blibelwzy.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.45uqqpgmj7e8vfg9.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.48q26q968g09fk0z.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.48x3fqbpy14aatae.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4a4xoe9srx9e1587.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4ae35dw0bjjekbps.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4dlhfj9c443k50sc.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4e00zxlnq4onq5xr.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4gfxnda0gap5oexf.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4hi7etsu61n8nhp8.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4kut7e2sotq1xjaa.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4l0zk8d6wvz3h9d1.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4linnu96q48awi4e.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4o2grl3rv7qbbsfg.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4p0qwsu7h1ors9zz.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4te38wxbw1m2wv4k.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4tvwngd82o62rqaa.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.4zxy2hj8ydnrvu3v.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.52kj71kyi82q5qwf.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.52zvi59mix4fxhj.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.55dmzvjygr9bvf3v.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.56a5u59godfxhrfw.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/cargo_flash-fe8889690c32bf94.58m7y5gswzaqgrxn.rcgu.o" "/home/ellie/Projects/cargo-flash/target/debug/deps/ca [... cut ...]
ii  libftdi1-dev                                1.5-5build1                                  amd64        Development files for libftdi1
ii  libusb-1.0-0-dev:amd64                      2:1.0.24-3                                   amd64        userspace USB programming library development files
ii  libusb-dev                                  2:0.1.12-32build2                            amd64        userspace USB programming library development files
```

**Additional context**
Add any other context about the problem here.

Tiwalun, 2021-12-11:
Thanks for reporting this!

The problem seems to be caused by the `libusb-dev` package, which causes problems with `rusb`.

`libusb-1.0-0-dev` by itself should be enough for probe-rs, can you please try removing `libusb-dev` and checking if that fixes it for you?

sepotvin, 2021-12-15:
I had the exact same problem and the proposed solution (remove libusb-dev) fixed it for me. Thanks!

linuxtim, 2021-12-20:
Here's a possible documentation fix for this.

n.b. upstream bug: https://github.com/a1ien/rusb/issues/117

and downstream bug: https://github.com/knurling-rs/probe-run/issues/289

Yatekii, 2022-01-29:
Fixed in #212 :)

closed 2022-01-29 by Yatekii